cbrake/linux-serial-test #11: Error setting RS-232 mode: Inappropriate ioctl for device

DiDu0815, 2019-03-15:
First, thanks for the code, it is very useful. I use it on Ubuntu 16.04.
On each run, regardless of parameters used, I get the message above. The code seems to work well however; I can verify it using a scope attached to RX-TX on the loopback plug.
Simple example to produce the message: **"sudo ./linux-serial-test -p /dev/ttyS0"**.
The message ist not in the source code, so it must be thrown from the includes. Any idea where this comes from an how to avoid it?

andy-shev, 2019-03-15:
One of the IOCTLs is not supported by your system.
I would suspect this one:

	ioctl(_fd, TIOCGRS485, &rs485)

cbrake, 2019-03-15:
there may be some way to detect if the platform supports RS485, and then only call those ioctls if appropriate.

DiDu0815, 2019-03-15:
True, the system has RS-232 only. So if any message in this case the "error" should be better something more like "informational", to avoid causing concerns at user side.

cbrake, 2019-10-29:
pull requests are welcome.

closed 2019-10-29 by cbrake