cbrake/linux-serial-test #17: Use BOTHER to set the baudrate

yegorich, 2019-10-29:
Use the modern way to set the baudrate using an integer and not a macro.

```
struct termios2 tio;

ioctl(fd, TCGETS2, &tio);
tio.c_cflag &= ~CBAUD;
tio.c_cflag |= BOTHER;
tio.c_ispeed = 12345;
tio.c_ospeed = 12345;
ioctl(fd, TCSETS2, &tio);

```
For more information see: https://stackoverflow.com/a/19992472/1113139

cbrake, 2019-10-29:
I like the idea of adding this code.

Is there any risk this new API won't work on some older systems? If so, should we offer both, default to the new, and perhaps have a command line option to force the old method if needed?

yegorich, 2019-10-29:
That's a good question. Adding support can be a little bit tricky. See this [note](https://github.com/npat-efault/picocom/blob/master/termios2.txt).

hackerb9, 2025-08-05:
I implemented this in [my fork](https://github.com/hackerb9/linux-serial-test) by putting the termios2 calls in a separate file to prevent the clash between `<asm/termios.h>` and `<termios.h>`. For the conflict with ioctls.h, I simply copied the prototype of ioctl() directly into the code. 

It works well with the FTDI and PL2303 drivers, even for non-genuine Future Tech chips. The MCT U232 driver, on the other hand, can only use the canonical baudrates.

andy-shev, 2025-08-08:
picocom developer has a good wrap-up about BOTHER (and IIRC I have already mentioned it in this project somewhere -- another PR or issue): https://github.com/npat-efault/picocom/blob/master/termios2.txt. So, please study that case before implementing it here.

hackerb9, 2025-08-08:
> picocom developer has a good wrap-up about BOTHER (and IIRC I have already mentioned it in this project somewhere -- another PR or issue): https://github.com/npat-efault/picocom/blob/master/termios2.txt. So, please study that case before implementing it here.

Thanks. Yes, I read that.