cbrake/linux-serial-test #22: PL2303HXD issues with high baud rates

swiftgeek, 2019-11-30:
6000000 and 12000000 have issues with linux-serial-test, but work fine in tio/picocom
```
linux-serial-test -b 6000000 -p /dev/ttyUSB0 -s -e
Linux serial test app
NOTE: non standard baud rate, trying custom divisor
zsh: floating point exception (core dumped)
linux-serial-te[631304] trap divide error ip:556b5c37a677 sp:7fffab836d80 error:0 in linux-serial-test[556b5c37a000+2000]
```

andy-shev, 2019-11-30:
Seems duplicate of #17.

cbrake, 2019-11-30:
does anyone happen to know what API the tio/picocom apps use? Might be able to learn something there.

@swiftgeek what platform are you testing on?

swiftgeek, 2019-12-01:
Archlinux with 5.3.x kernel

And BOTHER search on picocom repo returns https://github.com/npat-efault/picocom/blob/master/termios2.txt

cbrake, 2019-12-03:
Looking at the code a bit:

* https://github.com/npat-efault/picocom/blob/master/termios2.c#L159
* https://github.com/tio/tio/blob/master/src/setspeed2.c#L26
* https://github.com/tio/tio/blob/master/src/setspeed2.c#L26
* https://github.com/tio/tio/blob/master/src/tty.c#L609
* https://github.com/tio/tio/blob/master/configure.ac#L88

Seems we can do something like the tio project to conditionally use the new API.

d-el, 2025-02-27:
kernel: 6.11.0-17-generic

./linux-serial-test -s -e -p /dev/ttyACM0 -b 3100000
Linux serial test app
NOTE: non standard baud rate, trying custom divisor
Floating point exception (core dumped)

andy-shev, 2025-02-28:
> kernel: 6.11.0-17-generic
> 
> ./linux-serial-test -s -e -p /dev/ttyACM0 -b 3100000 Linux serial test app NOTE: non standard baud rate, trying custom divisor Floating point exception (core dumped)

Linux kernel doesn't use floating arithmetics (or math co-processor in general), so this exception comes from user space.  But I don't see anything in the linux-serial-test either. Do you have an update of glibc? When it was working last time what the kernel version was that? Also check other core libraries.