cbrake/linux-serial-test #27: The application hangs at tcdrain(_fd) if -c , -o and -i is used.

detari5365, 2020-02-11:
If I issue this command below, the application never exited. If I replaced tcdrain(_fd) with close(_fd), it does work.

**linux-serial-test  -s -e -p /dev/ttyS4 -b 4000000 -c -o 10 -i 10**

https://github.com/cbrake/linux-serial-test/blob/551bff2397aa5d9db0820e76c7e228bd43ccbe4e/linux-serial-test.c#L743

andy-shev, 2020-02-11:
What output did you see on the screen? What driver in use for this test? Do you have RTS/CTS lines actually connected?

detari5365, 2020-02-12:
Below is the output:
**Linux serial test app
Error setting RS-232 mode: Inappropriate ioctl for device
/dev/ttyS4: count for this session: rx=57344, tx=65520, rx err=0
/dev/ttyS4: TIOCGICOUNT: ret=0, rx=159734, tx=163815, frame = 0, overrun = 0, parity = 0, brk = 0, buf_overrun = 0
Stopped transmitting.
Stopped receiving.**

The driver is "dw-apb-uart".

Yes, RTS/CTS lines are connected. It is a loopback test.

cbrake, 2020-02-12:
@detari5365 I have not been able to reproduce this with a FTDI USB serial port with Rx/Tx looped back:

```
[cbrake@mars linux-serial-test]$ ./linux-serial-test -s -e -p /dev/ttyUSB2 -b 115200 -o 10 -i 10
Linux serial test app
/dev/ttyUSB2: count for this session: rx=58032, tx=59904, rx err=0
/dev/ttyUSB2: TIOCGICOUNT: ret=0, rx=278961, tx=281856, frame = 0, overrun = 0, parity = 0, brk = 0, buf_overrun = 0
Stopped transmitting.
Stopped receiving.
/dev/ttyUSB2: count for this session: rx=104160, tx=105984, rx err=0
/dev/ttyUSB2: TIOCGICOUNT: ret=0, rx=326577, tx=329216, frame = 0, overrun = 0, parity = 0, brk = 0, buf_overrun = 0
```

I'm not sure why the counts to match at the end of the session -- seems like they should, but perhaps we are only waiting until tx is drained, and not all the data has been received yet. 

So, if I increase -i to 15, then all the data comes in:

```

[cbrake@mars linux-serial-test]$ ./linux-serial-test -s -e -p /dev/ttyUSB2 -b 115200 -o 10 -i 15
Linux serial test app
/dev/ttyUSB2: count for this session: rx=65472, tx=69120, rx err=0
/dev/ttyUSB2: TIOCGICOUNT: ret=0, rx=502641, tx=505856, frame = 0, overrun = 0, parity = 0, brk = 0, buf_overrun = 0
Stopped transmitting.
/dev/ttyUSB2: count for this session: rx=115200, tx=115200, rx err=0
/dev/ttyUSB2: TIOCGICOUNT: ret=0, rx=552369, tx=555008, frame = 0, overrun = 0, parity = 0, brk = 0, buf_overrun = 0
Stopped receiving.
/dev/ttyUSB2: count for this session: rx=115200, tx=115200, rx err=0
/dev/ttyUSB2: TIOCGICOUNT: ret=0, rx=552369, tx=555008, frame = 0, overrun = 0, parity = 0, brk = 0, buf_overrun = 0
```

Seems like tcdrain should be supported:

[image]

Is it possible that the dw-apb-uart driver is not complete?

detari5365, 2020-02-12:
@cbrake ,
At my side, if I increase -i to 15, the application able to exit.

**./linux-serial-test -s -e -p /dev/ttyS4 -b 115200 -c -o 10 -i 15
Linux serial test app
Error setting RS-232 mode: Inappropriate ioctl for device
/dev/ttyS4: count for this session: rx=57344, tx=65520, rx err=0
/dev/ttyS4: TIOCGICOUNT: ret=0, rx=262157, tx=266215, frame = 0, overrun = 0, parity = 0, brk = 0, buf_overrun = 0
Stopped transmitting.
/dev/ttyS4: count for this session: rx=106470, tx=106470, rx err=0
/dev/ttyS4: TIOCGICOUNT: ret=0, rx=311283, tx=311260, frame = 0, overrun = 0, parity = 0, brk = 0, buf_overrun = 0
Stopped receiving.
/dev/ttyS4: count for this session: rx=106470, tx=106470, rx err=0
/dev/ttyS4: TIOCGICOUNT: ret=0, rx=311283, tx=311260, frame = 0, overrun = 0, parity = 0, brk = 0, buf_overrun = 0**

cbrake, 2020-02-12:
@detari5365 I suspect there are still issues with the dw-apb-uart driver unless I'm not understanding what `tcdrain()` does ...

cbrake, 2020-02-15:
closing this -- please re-open if you feel there is still an issue we need to resolve in this app.

closed 2020-02-15 by cbrake