cbrake/linux-serial-test #29: Baud B38400 cannot be set after non-standard baud was used

liubiin, 2020-04-13:
If first set a non-standard baud rate, for example, 128000, get_baud() returns -1, then ASYNC_SPD_CUST flag is set in kernel driver because of set_baud_divisor(), but it is never got cleared.

Then later if you want to use standard 38400 baud rate, the kernel driver cannot work properly because user space didn't ask kernel to clear ASYNC_SPD_CUST flag in the first scenario.

What is the proper fix in this code to clear ASYNC_SPD_CUST for the kernel after use?

Note, this scenario has nothing to do with the cmdline custom divisor parameter.

liubiin, 2020-04-14:
The following patch works, though I don't like it.

```
diff --git a/linux-serial-test.c b/linux-serial-test.c
index 6ff21f43c723..4277f69ed465 100644
--- a/linux-serial-test.c
+++ b/linux-serial-test.c
@@ -124,6 +124,25 @@ static void set_baud_divisor(int speed, int custom_divisor)
 	}
 }
 
+static void clear_custom_speed_flag()
+{
+	struct serial_struct ss;
+	if (ioctl(_fd, TIOCGSERIAL, &ss) < 0) {
+		perror("TIOCGSERIAL failed");
+		exit(1);
+	}
+
+	if ((ss.flags & ASYNC_SPD_MASK) != ASYNC_SPD_CUST)
+		return;
+
+	ss.flags &= ~ASYNC_SPD_MASK;
+
+	if (ioctl(_fd, TIOCSSERIAL, &ss) < 0) {
+		perror("TIOCSSERIAL failed");
+		exit(1);
+	}
+}
+
 // converts integer baud to Linux define
 static int get_baud(int baud)
 {
@@ -589,6 +608,8 @@ int main(int argc, char * argv[])
 		set_baud_divisor(_cl_baud, _cl_divisor);
 	} else {
 		setup_serial_port(baud);
+		if (baud == B38400)
+			clear_custom_speed_flag();
 	}
 
 	set_modem_lines(_fd, _cl_loopback ? TIOCM_LOOP : 0, TIOCM_LOOP);

```

cbrake, 2020-04-14:
would it make more sense to always clear the custom_speed_flag() if we are not doing a custom baud -- IE do we need to check if baud == B38400?

liubiin, 2020-04-14:
Probably yes. It doesn't matter to the uart driver, since the driver should check for (baud == B38400 && flags & SPD_CUST) when handling custom divisor.

cbrake, 2020-04-14:
in general, this seems like a good improvement to me -- makes the app a little more robust.

liubiin, 2020-04-14:
I am looking for a better solution, for example, should clear the flag at the end before close the port instead?

cbrake, 2020-04-14:
yeah, would be good to clean up after ourselves, but I don't think it hurts to make sure the port is initialized correctly as well.

liubiin, 2020-04-14:
So do you want to me to send the patch above in a PR?

cbrake, 2020-04-14:
yes, I think it is a good change, so I'll merge PR.

closed 2020-04-14 by cbrake