cbrake/linux-serial-test #32: Parameters to test RS485

amankumar1194, 2020-10-19:
From the options available, I see "-q" to enable the RS485 direction control. But I dont under what all the parameters can be passed to validate the working of RS485 driver using your app. 

If someone can help me with the proper parameters to test RS485 with this app then it would be very much helpful.

cbrake, 2020-10-20:
To test RS485, generally you need two ports, wire them together, and then send data from one port to the other, and then back the other way. You could run one port with transmit disabled, and configure the other instance to send data. This app uses a counting pattern to detect missed data.

andy-shev, 2023-01-23:
FYI: Linux kernel gained 9-bit mode and RS422 support (https://elixir.bootlin.com/linux/v6.2-rc5/source/include/uapi/linux/serial.h#L130), maybe someone wants to implement this in the testing tool...

htot, 2023-01-23:
422? This one `SER_RS485_RX_DURING_TX	- Full-duplex RS485 line.`?

andy-shev, 2023-02-02:
> 422? This one `SER_RS485_RX_DURING_TX - Full-duplex RS485 line.`?

It's a confusion in the Linux kernel. Some of the drivers use that to set RS422 compatibility  in the registers of a certain hardware. But yes, RS485 in some cases can be full-duplex (needs two differential pairs, or the specific methods to communicate between peers on the physical level (I don't remember what Synopsis is ptoviding).

htot, 2023-02-03:
I think the latter is a confusion by some manufacturers calling their RS422 a full-duplex RS485 (because they used RS485 drivers in the design).

What confused me even more is that their exists RS422 in a multiple slave configuration.