cbrake/linux-serial-test #33: Ubuntu 20.04.1 LTS 64bit getting: Error setting RS-232 mode: Inappropriate ioctl for device

oldunixguy, 2020-10-28:
I compiled the src with gcc -o linux-serial-test linux-serial-test.c
I have a null modem cable from ttyS0 to ttyS4. Both tty are real rs-232 serial ports.

In shell1 I ran:
linux-serial-test -s -e -p /dev/ttyS4 -t -c -l 250 -b 57600
In shell2 I ran:
linux-serial-test -s -e -p /dev/ttyS0 -r -c -b 57600

In both instances I get:
Error setting RS-232 mode: Inappropriate ioctl for device

and then both instances appear to operate normally.
The error output is paltry. Any suggestions on what to add to the output to narrow down for you what is wrong?

thanks
oldunixguy

cbrake, 2020-10-28:
It appears we are trying to clear the RS485 settings on a system that does not support it:

https://github.com/cbrake/linux-serial-test/blob/master/linux-serial-test.c#L569

Can you try this branch and see if that works?

https://github.com/cbrake/linux-serial-test/tree/feature-fix-rs485-error

All, any concerns with above change?

oldunixguy, 2020-10-28:
roger that. I've been writing C since 1981 so whatever you need I can do
it....
while in there would you consider me "enhancing" the error outputs
throughout, such as adding the src line number the error emits from?
thanks
rich

cbrake, 2020-10-28:
sure, pull requests are welcome

oldunixguy, 2020-10-29:
havnt done a pull request.... I obtained the software using the create a
zip button.
do you have a crib sheet on what I should do?
thanks
rich

oldunixguy, 2020-10-29:
I am reading https://guides.github.com/activities/hello-world/ and will
give this a try...
rich

oldunixguy, 2020-10-29:
Cliff,

I got the version you posted. I'm still new to git and couldnt figure out
how to create a branch on your branch....
I added my enhanced perror and fprintf(stderr...) to your version and then
ran the test.... I get the same error at the same (relative ) location in
the code.
It still reports:
setup_serial_port() 581: Error setting RS-232 mode: Inappropriate ioctl for
device

where 581 is the line number....

Please bear with me while I figure out how to use git... I'm an old SCCS
guy!

To that end I attached my file version....
I'm going to start adding some more debugging and see where this takes me.

thanks
rich

oldunixguy, 2020-10-29:
Cliff,

here is the output of the first ioctl to get the rs485 params.... i
modified the code to:
ret = ioctl(_fd, TIOCGRS485, &rs485);
ERROR("ioctl ret = %d, errno %d\n", ret, errno);

setup_serial_port() 561: ioctl ret = 0, errno 0
rs485 ioctl output:
0x00000000 flags
0x00000000 delay_rts_before_send
0x00000000 delay_rts_after_send
0x00000000 padding[0]
0x00000000 padding[1]
0x00000000 padding[2]
0x00000000 padding[3]
0x00000000 padding[4]

This suggests that ubuntu 20 returns success to the TIOCGRS485 BUT the
contents of the params are all 0
Hummm.
rich

cbrake, 2020-10-29:
Looks like you are making good progress. The IOCTL implementation is likely a function fo the Linux kernel serial driver, which can vary from system to system. 

I went ahead and merged my changes to master: https://github.com/cbrake/linux-serial-test/pull/34

Figuring out Git would be good -- if you need any help with Git, feel free to [contact me directly](https://github.com/cbrake) via email. Figuring out the command line git client is the best approach (vs making edits in the GitHub web UI).