cbrake/linux-serial-test #38: write stuck in while(1)

Memarnejad, 2021-12-28:
Hi,
In function `process_write_data`, repeat condition may lead to `while(1)`.
Suppose that user does not specify _--tx-bytes_:
`int repeat = (_cl_tx_bytes == 0);`
 so the repeat condition will be 1 and if the following condition is not satisfied, we will stuck in while(1)
`if (c < actual_write_size) {
	_write_count_value = _write_data[c];
	repeat = 0;
}`
If driver can send all data, then `c = actual_write_size` which leads the condition not being satisfied and we stuck in while(1)

But the question is that why does it work with some devices?
I guess that's because drivers can't send 1024bytes in **ONE** round (the remaining will be send in the next round ... handled in userspace) and the `c < actual_write_size` is satisfied and we don't stuck in while(1). But what if the driver can send 1024bytes? Unfortunately, it will get stuck in while(1)

I fixed this issue by simply replacing **<** with **<=**. Here is the modification:
`if (c <= actual_write_size) {
	_write_count_value = _write_data[c];
	repeat = 0;
}`

Goshik92, 2024-04-10:
Hi @Memarnejad, I have the same issue, and it looks like the solution you proposed fixes it. Please consider opening a PR for merging your fix into the main branch.

andy-shev, 2024-04-11:
FIWI, It's not uncommon that driver **can** send even more than 1kB at once with a help of DMA.