cbrake/linux-serial-test #49: Sending custom data over UART

karthikkumarb, 2023-10-02:
Hi, 

I need to send some custom data like "\x00\x13\x00" over UART using this code, Can you please help me how to send some custom data?

Thanks,
Karthik

andy-shev, 2024-05-20:
Basic idea is to have the port to be set in a `raw` mode. In this case the TTY layer won't mangle any character on wire (it might be an exception fox XON/XOFF case, but anyway it's highly recommended to have a HW flow control).

P.S. I dunno why you need this code to be used, but in shell the `printf ...` will do the job.

cbrake, 2024-05-20:
You can also use the following in a various programming languages:

- https://pyserial.readthedocs.io/en/latest/pyserial.html
- https://pkg.go.dev/go.bug.st/serial

Or simply modify this program to put your data in the buffer that gets sentout.

closed 2024-05-20 by cbrake