cbrake/linux-serial-test #55: Use different ports for RX and TX

Harvie, 2024-01-03:
Hello,
i would like to use this to test connection between two serial ports (instead of just using loopback). can you please add such functionality?

Usecase:  i have two rs485 ports. if there is bad connection between rs485 line driver and connector, the loopback will work anyway, because rs485 uses single half duplex line for RX and TX. therefore i need to connect two separate ports together to test it properly.

cbrake, 2024-01-03:
Can you run two instances of the -- one on each port?

Harvie, 2024-01-04:
Will that work out of the box? two instances facing each other?

cbrake, 2024-01-04:
It's been a long time since I've done this -- but I think if one instance generates a known counting pattern, and the other instance is looking for the same pattern, it should be OK. Give it a try and let us know how it goes ...

andy-shev, 2024-05-20:
Just a note: The first one to start is Rx side (for obvious reasons).

cbrake, 2024-05-20:
I think the solution proposed is reasonable, closing ...

closed 2024-05-20 by cbrake