cbrake/linux-serial-test #59: Does linux-serial-test tool support sending data with a specific pattern?

snowuyl, 2025-01-24:
I would like to use linux-serial-test tool to send UART data with pattern 0xa5, 0x0, 0x52, 0x1, 0x52, 0x40. Does linux-serial-test tool support sending data with a specific pattern?

andy-shev, 2025-01-27:
As far as you can see in the documentation the custom pattern can be only a byte or two bytes. You need to add yourself the necessary functionality (hence this issue is "feature request"). Add a new switch to use a file for that, add necessary file handling (using `open()` + `mmap()` + `close()`) and use that memory as repetitive data to send. One day work or so. Everybody will appreciate your work!

snowuyl, 2025-01-27:
Thanks for your reply!

closed 2025-01-27 by snowuyl